flipperdevices/qFlipper #183: Arm64 linux

d3k4-k3rb3r0s, 2023-05-09:
Qflipper doesnt install on arm64 Linux. And there's no way I'm using the walking surveillance bot called Windows ten... so what gives... I'm sitting here with a brand new flipper wondering why a hacking device is made only to be compatible with daily driver linux and windows. When is this going to be fixed?

gsurkov, 2023-05-09:
The quickest way would be to compile iqFlipper from sources on your arm64 system, there should be no source-level limitations for that. Build instructions can be found [here](https://github.com/flipperdevices/qFlipper#standalone-build). Please open another issue if you decide to go this route and encounter any problems.

Secondly, there may already exist pre-built packages for the distribution you are using, and if not, you may ask its maintainers to create one or even contribute your own package.

Official binaries for arm64 Linux systems will be added only if there will be enough demand.

closed 2023-05-09 by gsurkov

Silent-Hunter, 2023-09-27:
How will you know if there's enough demand? I'm in the same situation, my daily driver is a Raspberry Pi.