flipperdevices/qFlipper #185: APP inst QFLIPPER

SaintBufd, 2023-05-26:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
What were you trying to do?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
A clear and concise description of what happened instead.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
2. Does reconnecting/rebooting Flipper solve it?

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
2. Right click on the log window.
3. In the context menu, select "Browse all logs...".
4. A folder containing several log files should open. 
5. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.

zhovner, 2023-05-26:
Spam?

closed 2023-05-26 by zhovner

SaintBufd, 2023-05-26:
when I download the qflipper app instead I get a 3D printer app.