flipperdevices/qFlipper #189: qFlipper Appimage won't load

mmiller7, 2023-06-18:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
What were you trying to do?
Run qFlipper Appimage on Linux Mint 21.1

**To Reproduce**
Steps to reproduce the behavior:
1. Download qFlipper
2. Add executable permissions
3. Run it

NOTE: Running by double-clicking does nothing (not work, no error).  Running in terminal shows an error and exits.

**Expected behavior**
I expect it to open

**Actual behavior**
```
./qFlipper-x86_64-1.3.2.AppImage
basename: extra operand ‘Downloads/qFlipper-x86_64-1.3.2.AppImage’
Try 'basename --help' for more information.

sudo ./qFlipper-x86_64-1.1.3.AppImage 
basename: extra operand ‘Downloads/qFlipper-x86_64-1.1.3.AppImage’
Try 'basename --help' for more information.
```

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? YES
5. Does reconnecting/rebooting Flipper solve it? NO

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).  APP DOESN'T OPEN, CAN'T GET TO THE BUTTON.
3. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
2. Right click on the log window.
6. In the context menu, select "Browse all logs...".
7. A folder containing several log files should open. 
8. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.

gsurkov, 2023-11-01:
This is caused by spaces being present in the path to the executable.
Put the AppImage file in a path with no spaces in it as a temporary workaround.

gsurkov, 2023-11-02:
Fixed in #201.

closed 2023-11-02 by gsurkov