flipperdevices/qFlipper #191: Ability to Resize Window

theundeadlich, 2023-07-13:
Could we have the ability to resize the qFlipper window for those of us on higher or different resolution screens so that we can see it better.

zhovner, 2023-07-14:
Can you please show how it looks on your screen?

theundeadlich, 2023-07-14:
Sure thing!

Here it is on 3840x2160

[image]

zhovner, 2023-07-16:
Hm, it should scale the window automatically, isnt? @gsurkov

theundeadlich, 2023-09-13:
I am able to make the application slightly bigger by using High DPI Scaling Override and setting it toi system, but this still doesn't allow us to rescale the window.

[image]

gsurkov, 2023-11-01:
TL;DR:
qFlipper does its best to guess what the window size should be according to the system settings. In case it fails to do so, it's possible to instruct it by setting the environment variable `QT_SCALE_FACTOR` to what feels best. Make sure not to set it globally as it will affect other Qt-based applications, too.

A bit more trivia:
1. qFlipper uses a fixed-size window by design. It was a deliberate decision to make it look like the janky early 2000s programs with custom windows. We might rethink this decision for the version 2.0 as it causes more problems than anticipated (people have stopped designing apps this way for a reason).
2. qFlipper reads HiDPI settings from the underlying OS and adjusts automatically thanks to the Qt library. It doesn't look at the resolution or physical screen size, but at the scaling ratio set in the system. Might add an explicit setting if it doesn't work well automatically.

closed 2023-11-01 by gsurkov