flipperdevices/qFlipper #194: Permission Denied Error on version 1.3.0 and 1.3.2 when connecting flipper with qFlipper

hack-wizard, 2023-07-30:
**Describe the bug**
qFlipper fails to connect on Debian 11 over USB, I've tried the old version (which seems to have stopped working and latest release 1.3.2

**To Reproduce**
Steps to reproduce the behavior:
1. Run qFlipper as root (sudo) via terminal (also ran the rules install command first, it confirmed they were)
2. Ensure Bluetooth is off on flipper
3. Connect flipper from any USB port
4. I get the permission denied image
5. Terminal shows it repeatedly tried to connect to serial port at /dev/ttyACM0
6. If I click READ MORE I get an error saying Parameter "link' is not declared

**Expected behavior**
The flipper properly connects and I can update firmware/manipulate the filesystem

**Actual behavior**
It acts as though it cannot figure out how to initiate the serial connection

**Additional troubleshooting**
1. I rebooted, tried different cables and ports and even closed all browser windows.
2. using command `screen /dev/ttyACM0 115200` connects over the flippers serial port, proving the flipper itself isn't the issue
3. Also tried reinstalling the rules and confirming my user is in the dialout group

See log below:
603 [APP] qFlipper version 1.3.2 commit 7b0839e1 2023-06-08T03:58:22
603 [APP] OS info: Debian GNU/Linux 11 (bullseye) 11 5.10.0-15-amd64 Qt 6.4.3
617 [REG] Detected new device: VID_0x483:PID_0x5740
795 [DBG] Trying serial port  at /dev/ttyACM0
823 [DBG] Trying serial port  at /dev/ttyACM0
855 [DBG] Trying serial port  at /dev/ttyACM0
888 [DBG] Trying serial port  at /dev/ttyACM0
920 [DBG] Trying serial port  at /dev/ttyACM0
957 [DBG] Trying serial port  at /dev/ttyACM0
988 [DBG] Trying serial port  at /dev/ttyACM0
1019 [DBG] Trying serial port  at /dev/ttyACM0
1048 [DBG] Trying serial port  at /dev/ttyACM0
1070 [DBG] Trying serial port  at /dev/ttyACM0
1092 [DBG] Trying serial port  at /dev/ttyACM0
1114 [DBG] Trying serial port  at /dev/ttyACM0
1136 [DBG] Trying serial port  at /dev/ttyACM0
1158 [DBG] Trying serial port  at /dev/ttyACM0
1159 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1159 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1181 [DBG] Trying serial port  at /dev/ttyACM0
1210 [DBG] Trying serial port  at /dev/ttyACM0
1240 [DBG] Trying serial port  at /dev/ttyACM0
1261 [DBG] Trying serial port  at /dev/ttyACM0
1284 [DBG] Trying serial port  at /dev/ttyACM0
1306 [DBG] Trying serial port  at /dev/ttyACM0
1329 [DBG] Trying serial port  at /dev/ttyACM0
1352 [DBG] Trying serial port  at /dev/ttyACM0
1373 [DBG] Trying serial port  at /dev/ttyACM0
1395 [DBG] Trying serial port  at /dev/ttyACM0
1426 [DBG] Trying serial port  at /dev/ttyACM0
1460 [DBG] Trying serial port  at /dev/ttyACM0
1491 [DBG] Trying serial port  at /dev/ttyACM0
1522 [DBG] Trying serial port  at /dev/ttyACM0
1560 [DBG] Trying serial port  at /dev/ttyACM0
1591 [DBG] Trying serial port  at /dev/ttyACM0
1623 [DBG] Trying serial port  at /dev/ttyACM0
1660 [DBG] Trying serial port  at /dev/ttyACM0
1691 [DBG] Trying serial port  at /dev/ttyACM0
1713 [DBG] Trying serial port  at /dev/ttyACM0
1742 [DBG] Trying serial port  at /dev/ttyACM0
1772 [DBG] Trying serial port  at /dev/ttyACM0
1801 [DBG] Trying serial port  at /dev/ttyACM0
1832 [DBG] Trying serial port  at /dev/ttyACM0
1861 [DBG] Trying serial port  at /dev/ttyACM0
1891 [DBG] Trying serial port  at /dev/ttyACM0
1921 [DBG] Trying serial port  at /dev/ttyACM0
1957 [DBG] Trying serial port  at /dev/ttyACM0
1986 [DBG] Trying serial port  at /dev/ttyACM0
2016 [DBG] Trying serial port  at /dev/ttyACM0
2046 [DBG] Trying serial port  at /dev/ttyACM0
2069 [DBG] Trying serial port  at /dev/ttyACM0
2091 [DBG] Trying serial port  at /dev/ttyACM0
2122 [DBG] Trying serial port  at /dev/ttyACM0
2160 [DBG] Trying serial port  at /dev/ttyACM0
2190 [DBG] Trying serial port  at /dev/ttyACM0
2219 [DBG] Trying serial port  at /dev/ttyACM0
2260 [DBG] Trying serial port  at /dev/ttyACM0
2295 [DBG] Trying serial port  at /dev/ttyACM0
2329 [DBG] Trying serial port  at /dev/ttyACM0
2363 [DBG] Trying serial port  at /dev/ttyACM0
2393 [DBG] Trying serial port  at /dev/ttyACM0
2424 [DBG] Trying serial port  at /dev/ttyACM0
2461 [DBG] Trying serial port  at /dev/ttyACM0
2491 [DBG] Trying serial port  at /dev/ttyACM0
2521 [DBG] Trying serial port  at /dev/ttyACM0
2551 [DBG] Trying serial port  at /dev/ttyACM0
2580 [DBG] Trying serial port  at /dev/ttyACM0
2610 [DBG] Trying serial port  at /dev/ttyACM0
2640 [DBG] Trying serial port  at /dev/ttyACM0
2671 [DBG] Trying serial port  at /dev/ttyACM0
2701 [DBG] Trying serial port  at /dev/ttyACM0
2731 [DBG] Trying serial port  at /dev/ttyACM0
2762 [DBG] Trying serial port  at /dev/ttyACM0
2792 [DBG] Trying serial port  at /dev/ttyACM0
2822 [DBG] Trying serial port  at /dev/ttyACM0
2857 [DBG] Trying serial port  at /dev/ttyACM0
2886 [DBG] Trying serial port  at /dev/ttyACM0
2918 [DBG] Trying serial port  at /dev/ttyACM0
2950 [DBG] Trying serial port  at /dev/ttyACM0
2982 [DBG] Trying serial port  at /dev/ttyACM0
3014 [DBG] Trying serial port  at /dev/ttyACM0
3046 [DBG] Trying serial port  at /dev/ttyACM0
3078 [DBG] Trying serial port  at /dev/ttyACM0
3110 [DBG] Trying serial port  at /dev/ttyACM0
3142 [DBG] Trying serial port  at /dev/ttyACM0
3174 [DBG] Trying serial port  at /dev/ttyACM0
3205 [DBG] Trying serial port  at /dev/ttyACM0
3237 [DBG] Trying serial port  at /dev/ttyACM0
3268 [DBG] Trying serial port  at /dev/ttyACM0
3299 [DBG] Trying serial port  at /dev/ttyACM0
3329 [DBG] Trying serial port  at /dev/ttyACM0
3361 [DBG] Trying serial port  at /dev/ttyACM0
3393 [DBG] Trying serial port  at /dev/ttyACM0
3425 [DBG] Trying serial port  at /dev/ttyACM0
3461 [DBG] Trying serial port  at /dev/ttyACM0
3492 [DBG] Trying serial port  at /dev/ttyACM0
3524 [DBG] Trying serial port  at /dev/ttyACM0
3562 [DBG] Trying serial port  at /dev/ttyACM0
3593 [DBG] Trying serial port  at /dev/ttyACM0
3625 [DBG] Trying serial port  at /dev/ttyACM0
3660 [DBG] Trying serial port  at /dev/ttyACM0
3693 [DBG] Trying serial port  at /dev/ttyACM0
3726 [DBG] Trying serial port  at /dev/ttyACM0
3762 [DBG] Trying serial port  at /dev/ttyACM0
3778 [REG] Device initialization failed: Failed to find a suitable serial port
5183 [qt.qml.context] qrc:/components/TextBox.qml:39:9 Parameter "link" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
5183 [default] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
13384 [default] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

Also see screenshot:
[image]

tinkercnc, 2023-08-03:
Put this 
`SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0666", GROUP="plugdev"`
in a file like 
  `99-flipperz.rules`
under
  `/etc/udev/rules.d`
and restart udev with
  `sudo udevadm control -R`
.
Add your user to the group plugdev.

hack-wizard, 2023-08-04:
Created that file, added what was requested, restarted udev, added user to the group and... that had no impact.

Should I have added this line to 42-flipperzero.rules by any chance?  That's what I have in the directory specified

gsurkov, 2023-11-01:
@hack-wizard Are you using the AppImage version or a package from another source?