flipperdevices/qFlipper #196: connect via bluetooth

nanderer, 2023-10-03:
hi,

it would be grate if i could use qFlipper not only using usb but also using bluetooth like from the phone, i know its slower but it should be possible.

Or is it possible to use other tools like bluetoothctl and rfcomm to make a serial port for the flipper that i then can use using qFlipper or with something else (like screen in serial terminal mode)?

Thanks in advance

zhovner, 2023-10-04:
If you use macOS and have Apple Silicon computer, you can run [Flipper Mobile App ](https://flpr.app) on your desktop.

nanderer, 2023-10-07:
yea. i sure could also get it somehow running in an android emulator (if i could hook one up that gives access to bluetooth hardware of my laptop (maybe if its usb using virtualbox and android-x86)) buts that not the point. I would prefer a native linux app that connects to the flipper using bluetooth...

gsurkov, 2023-11-01:
No, this functionality is not planned.

closed 2023-11-01 by gsurkov

Kabouik, 2023-11-07:
Just to bring grist to the mill (but I understand you decided already), qFlipper over Bluetooth would also allow using the Flipper without showing it, which could be useful to avoid raising eyebrows when trying to fool people with IR remotes or other pranks.

There are also mobile OSes that can install qFlipper (I just installed it on my phone running Droidian, and it has a keyboard, so I could even use that device to edit text files before loading them into my Flipper and controlling it, while the Flipper stays in my bag).

This would be quite handy from a laptop too, of course.