flipperdevices/qFlipper #197: Debian 12

FalsePhilosopher, 2023-10-09:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
Qflipper would not connect to the FZ on debian 12.

**To Reproduce**
Steps to reproduce the behavior:
Connect fz to qflipper on debian 12

**Expected behavior**
To connect or at least suggest the install rules commands instead of just saying it's broken and to restore the FW.

**Actual behavior**
Would not connect
**Additional troubleshooting**
Tried reboots, I fixed it by googling qflipper linux and found this blog post detailing the fix https://sysadmin102.com/2023/06/installing-qflipper-on-debian-linux/
```
cat << EOF | sudo tee /etc/udev/rules.d/42-flipperzero.rules
> #Flipper Zero serial port
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", ATTRS{manufacturer}=="Flipper Devices Inc.", TAG+="uaccess", GROUP="uucp"
#Flipper Zero DFU
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", ATTRS{manufacturer}=="STMicroelectronics", TAG+="uaccess", GROUP="uucp"
EOF
```
There should be more documentation on different linux distros and qflipper an what needs to be done to get qflipper working.
Here's my log anyway even though I fixed it.

580 [APP] qFlipper version 1.3.2 commit 7b0839e1 2023-06-08T04:58:22
580 [APP] OS info: Debian GNU/Linux 12 (bookworm) 12 6.1.0-10-amd64 Qt 6.4.3
1251 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1364 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
6783 [USB] Failed to open device: Access denied (insufficient permissions)
6803 [USB] Failed to open device: Access denied (insufficient permissions)
6824 [USB] Failed to open device: Access denied (insufficient permissions)
6844 [USB] Failed to open device: Access denied (insufficient permissions)
6865 [USB] Failed to open device: Access denied (insufficient permissions)
6885 [USB] Failed to open device: Access denied (insufficient permissions)
6905 [USB] Failed to open device: Access denied (insufficient permissions)
6926 [USB] Failed to open device: Access denied (insufficient permissions)
6946 [USB] Failed to open device: Access denied (insufficient permissions)
6967 [USB] Failed to open device: Access denied (insufficient permissions)
6987 [USB] Failed to open device: Access denied (insufficient permissions)
7007 [USB] Failed to open device: Access denied (insufficient permissions)
7028 [USB] Failed to open device: Access denied (insufficient permissions)
7049 [USB] Failed to open device: Access denied (insufficient permissions)
7069 [USB] Failed to open device: Access denied (insufficient permissions)
7090 [USB] Failed to open device: Access denied (insufficient permissions)
7110 [USB] Failed to open device: Access denied (insufficient permissions)
7131 [USB] Failed to open device: Access denied (insufficient permissions)
7151 [USB] Failed to open device: Access denied (insufficient permissions)
7172 [USB] Failed to open device: Access denied (insufficient permissions)
7192 [REG] Incomplete device info: VID_0x483:PID_0x5740
17728 [qt.qml.context] qrc:/components/TextBox.qml:39:9 Parameter "link" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.

gsurkov, 2023-11-01:
>There should be more documentation on different linux distros and qflipper an what needs to be done to get qflipper working.

There are instructions on setting up device permissions in [Readme.md](https://github.com/flipperdevices/qFlipper#linux-1) and the [official documentation](https://docs.flipper.net/qflipper). 

More so, qFlipper itself displays a (graphical) error message if it fails to open the serial port due to insufficient permissions.

Did they not help?

FalsePhilosopher, 2023-11-02:
As I said it did not show the install rules error message, but instead shows restore firmware error.

I027614, 2024-01-06:
Same happens with Mageia 9 (64 bits). It works with AppImage, though.

sur5r, 2025-12-10:
Debian Maintainer of qFlipper here. qFlipper is available both in bookworm (12) and trixie (13). Trixie even has the current release. The native package ships the necessary udev rules.