flipperdevices/qFlipper #216: Cant Update or erase my flipper

Hamadatail, 2024-03-26:
### Describe the bug.

i have been trying to update my flipper firmware for a while and can not do as its keep failing regardless what method you use! i have trued to reboot it or delete the sd card but no changes 
<img width="805" alt="Screenshot 2024-03-25 at 11 23 26" src="https://githu
[image]
b.com/flipperdevices/flipperzero-firmware/assets/115875741/bc32c0b5-b8cb-460c-9945-e219b9a22a4d">

### Reproduction

n

incyi, 2024-08-17:
Any update after 5 months ?

gsurkov, 2024-08-17:
@incyi Does this problem occur for you?
If yes, I would like to have some more information, such as:
- Logs
- Current firmware version on your Flipper Zero
- OS Type/Version

incyi, 2024-08-17:
No, I was just checking if @Hamadatail has any update.

closed 2024-09-10 by skotopes