flipperdevices/qFlipper #220: Qflipper fails to compile on M1 MBA

N-coder82, 2024-08-04:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
I tried to compile Qflipper on a ARM mac, I installed Flipper's QT and libusb, unlinked both, and linked the flipper versions.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the qflipper repo, recursively
2. open a terminal and open the qflipper repo
3. run `./build_mac.sh:`
4. See error

**Expected behavior**
I expected the app to start to compile normally.

**Actual behavior**
The compile script ended on an error:
```
Project ERROR: libusb-1.0 development package not found
make: *** [sub-3rdparty-qmake_all] Error 3
```

**Additional troubleshooting**
Yes, it presists, it does not apply to the flipper.

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
this log is the compile log, not the flipper log:
+ PROJECT=qFlipper
+ BUILD_DIRECTORY=build_mac
+ '[' -d .git ']'
+ git submodule update --init
++ uname -s
+ [[ Darwin != \D\a\r\w\i\n ]]
++ uname -m
+ [[ arm64 == \a\r\m\6\4 ]]
++ /opt/homebrew/bin/brew shellenv
+ eval 'export HOMEBREW_PREFIX="/opt/homebrew";
export HOMEBREW_CELLAR="/opt/homebrew/Cellar";
export HOMEBREW_REPOSITORY="/opt/homebrew";
export PATH="/opt/homebrew/bin:/opt/homebrew/sbin${PATH+:$PATH}";
[ -z "${MANPATH-}" ] || export MANPATH=":${MANPATH#:}";
export INFOPATH="/opt/homebrew/share/info:${INFOPATH:-}";'
++ export HOMEBREW_PREFIX=/opt/homebrew
++ HOMEBREW_PREFIX=/opt/homebrew
++ export HOMEBREW_CELLAR=/opt/homebrew/Cellar
++ HOMEBREW_CELLAR=/opt/homebrew/Cellar
++ export HOMEBREW_REPOSITORY=/opt/homebrew
++ HOMEBREW_REPOSITORY=/opt/homebrew
++ export PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/Library/Frameworks/Python.framework/Versions/3.11/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Applications/iTerm.app/Contents/Resources/utilities
++ PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/Library/Frameworks/Python.framework/Versions/3.11/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Applications/iTerm.app/Contents/Resources/utilities
++ '[' -z '' ']'
++ export INFOPATH=/opt/homebrew/share/info:/opt/homebrew/share/info:
++ INFOPATH=/opt/homebrew/share/info:/opt/homebrew/share/info:
+ brew --version
Homebrew 4.3.12
+ brew --prefix libusb_universal
/opt/homebrew/opt/libusb_universal
+ brew --prefix qt_universal
/opt/homebrew/opt/qt_universal
+ rm -rf build_mac
+ mkdir build_mac
+ cd build_mac
++ xcrun -show-sdk-path
+ qmake -spec macx-clang 'CONFIG+=release qtquickcompiler' -o Makefile ../qFlipper.pro 'QMAKE_APPLE_DEVICE_ARCHS=x86_64 arm64' -early QMAKE_DEFAULT_LIBDIRS=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib
Info: creating stash file /Users/[-]/Programming/Git Repositories/qFlipper/build_mac/.qmake.stash
+ make qmake_all
cd 3rdparty/ && /opt/homebrew/bin/qmake -o Makefile /Users/[-]/Programming/Git\ Repositories/qFlipper/3rdparty/3rdparty.pro -spec macx-clang 'CONFIG+=release qtquickcompiler' 'QMAKE_APPLE_DEVICE_ARCHS=x86_64 arm64' -early QMAKE_DEFAULT_LIBDIRS=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib
Project ERROR: libusb-1.0 development package not found
make: *** [sub-3rdparty-qmake_all] Error 3

gsurkov, 2024-08-05:
Have you installed the dependencies listed in the [Readme](https://github.com/flipperdevices/qFlipper/blob/dev/README.md#macos)?

N-coder82, 2024-08-05:
> Have you installed the dependencies listed in the [Readme](https://github.com/flipperdevices/qFlipper/blob/dev/README.md#macos)?

Yeah

gsurkov, 2024-08-09:
We weren't able to reproduce the issue.
The compilation was successful on Mac M1 with the following software installed:
- MacOS: Sonoma 14.6.1
- Xcode: 15.4
- Qt: 6.3.1 static (from Flipper tap)
- libusb: 1.0.26 (from Flipper tap)
- mdgbuild: 1.6.2

The following command sequence was used for building:

```sh
brew tap flipperdevices/homebrew-flipper
brew install libusb_universal
brew install qt_universal
pip install dmgbuild
./build_mac.sh
```

You can check your installed versions by running

```sh
pip freeze | grep dmgbuild
brew info qt_universal
brew info libusb
```

closed 2024-08-09 by gsurkov