flipperdevices/qFlipper #224: Add download server config for 3'rd party fw

gameking2011, 2024-09-24:
put an option to change the update download server for 3'rd party fw

gameking2011, 2024-10-04:
my idea is to, instead of compiling the update server with qFlipper do something like this in backend.cpp line 37:
change: m_firmwareUpdateRegistry(new FirmwareUpdateRegistry("https://update.flipperzero.one/firmware/directory.json", this))
to: m_firmwareUpdateRegistry(new FirmwareUpdateRegistry("server.txt".Contents, this))
or something, i don't code w/ the qt framework

skinjun1987, 2025-01-13:
it keep saying cannot connect

zhovner, 2025-01-13:
Why not use a web installer of your alternative firmware?

GameLord2011, 2025-01-13:
@skinjun1987 Can you find all refs to the firmware server that point to the directory.json, and change all of those to the suggested text-file

skinjun1987, 2025-01-14:
so i downloaded the update finally but it only worked through the app which is weird I tried it on my PC all day pretty much. im super new and I dont know the lingo even. I am more of the action side lol... I am trying to stay afloat I dont really know what i possess fully with this thing but i feel like the sky is the limit. how do i use it for super dummies?

GameLord2011, 2025-01-14:
:hurtrealbad: @skinjun1987 What ide are you using (vscode, vscodium, vim, microsoft notepad, one of the jetbrains ones, neovim, vi, ettc.)