flipperdevices/qFlipper #231: Interface not shown when launching under Wayland.

asunekants, 2025-02-03:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**

When launching the AppImage on Fedora 41 with KDE Plasma on Wayland; The app launches, but no interface is presented.
When launching on the same system with KDE Plasma on X11, everything works as expected.

**To Reproduce**
Launch the AppImage on Fedora 41 with KDE Plasma on Wayland.

**Expected behavior**
The app launches with a visible and usable interface.

**Actual behavior**
The app launches, but no visible interface appears.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?

 - Yes.

2. Does reconnecting/rebooting Flipper solve it?

 - No

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).

[qFlipper-20250203-150649.txt](https://github.com/user-attachments/files/18644288/qFlipper-20250203-150649.txt)

3. A screenshot of the application window (if it is a graphical issue).

 - No screenshot is possible, as the interface is in an invisible window:

[image]

TheSainEyereg, 2025-03-08:
Launching with `sudo` solves it, however that's still the problem when launching it normally

sashokbg, 2025-05-18:
Same problem here. I would be happy to understand why the GUI requires sudo.
I have already added the required udev rules that allow my user to use the serial device to avoid using sudo

infifox, 2025-10-20:
I'm having the same issue running on Pop_OS with COSMIC. Running with `sudo` doesn't help, as I then get an error that it can't connect to the display.